google/OctoPrint-LEDStripControl #1: Problem with gpio group

pettib, 2017-03-19:
I´m testing your project and had a working Octoprint. But when after running
"usermod -G gpio pi" it stops to work. It can´t connect to the printer.
Error message:
Changing monitoring state from 'Operational' to 'Offline'
Changing monitoring state from 'Offline' to 'Detecting serial port'
Serial port list: ['/dev/ttyACM0']
Connecting to: /dev/ttyACM0
Error while connecting to /dev/ttyACM0: 'Timeout'
Changing monitoring state from 'Detecting serial port' to 'Error: Failed to autodetect serial port, please set it manually.'
Failed to autodetect serial port, please set it manually.
Connection closed, closing down monitor

I tried it with two different octoprint installations and both gave me the same result.
I´m not so good at Linux to find this out, just that it has something with rights to do.

precision, 2017-03-20:
Ack, I'm sorry about that.  Totally right, I somehow missed the -a,  fixed  #ac56ce7b53226c5225d86ff9335a8d6a6e8cf149

precision, 2017-03-20:
You should be able to restore the defaults with a 'usermod -a -G sudo, dialout,video pi'

closed 2017-03-20 by precision

pettib, 2017-03-20:
Thanks for clarifying. :-)
Can you tell me witch MOSFets you are using to control the Ledstrip from gpio.
I tried with 30N06L but they are built to work with 5v logic so they wont work. Need to find some that work nice with 3.3v logic.

thatch, 2017-03-20:
Hi pettib, I'm the board designer.  It has DNM3023L (SOT-23 ~6A N-Channel) on it, but it depends on how much current draw your strip needs.  precision's strip draws about 300mA, so this is much more than we need.  I also went surface-mount to keep things small.  These are about 38 cents each from digikey.

Did you try your 30N06L though?  Since you're only driving say 1A, given the  V_GS_th of 1.0 to 2.5V I think it's possible that it will work on 3.3 without overheating.  Note Figure 3 in the datasheet and extrapolate another curve the same distance to the top-left.  I suspect the y-intercept will only be about 40 or 50 mOhm.