google/OctoPrint-LEDStripControl #4: Chaotic comands - only through relay

elfhelmp, 2017-04-24:
Hello,
I'm trying to configure a 12V RGB LED strip and it would not work when RGB pins are connected directly onto the RPI ZERO although I have common GND with the LED strip PSU, RPI is just running on a step-down converter (150W). They all stay ON whatever the commands I send. If I measure the GND pins from 5V PSU with the GND pins from 12V PSU they are common so I don't think the problem is there, I'm using special cable for powering RPI with 20AWG wires.

I use a 4 relay board powered at 5V same buck as RPI, pins connected on the board, numbered G12, R16, B18. GND taken straight from the 12V PSU. This setup partly works. And I say partly, because the commands I send via terminal are not the same as described in your guide. I respected the coloring of the cables from the strip to the RPI and triple checked that pin numbers are OK and correspond to the said color on the strip. Here is the [setup](http://tinypic.com/r/v30g8x/9)

If I send M150 R U B then the LEDs turn off.
             M150 R255 turns on Blue.
             M150 turns ON all colors.

I am out of ideas and this is the first time I'm controlling LEDs with raspberry so I'm a novice, please advise.

thatch, 2017-04-24:
From this:

> If I send M150 R U B then the LEDs turn off.
> M150 R255 turns on Blue.
> M150 turns ON all colors.

For validating what I think is happening, can you also try with "M150 R", "M150 U", "M150 B" -- each should be a distinct color.  Ideally red, green, and blue.  Can you report on what the colors are; I suspect your setup will look like aqua, violet, and yellow-green.  Then if you run "M150 R U" it should be blue, "M150 R B" should be green, and "M150 U B" should be red.

If that's the case, then your relays are working but the output is inverted.  Does your relay board have connections labeled "NC" and "NO"?  If so, switch to using the other.

elfhelmp, 2017-04-25:
M150 R - aqua
M150 U - violet
M150 B - yellow green
M150 R U - blue
M150 R B - green
M150 U B - red

You were right. After moving to NC contacts, the commands worked as they should have. So the relays have to be in active state in order to keep the LEDs off. When I shutdown the RPI all LEDs turn on, which means this is not a good solution either. So... how do other people connect their LED strips to get it going? Can you please point me to a valid tutorial or give me a little explanation? many thanks!

precision, 2017-04-26:
Most people connect their LEDs via mosfets. Something ala https://learn.adafruit.com/rgb-led-strips/usage  I'm using a custom board (OSHW release soon) designed by Tim.

elfhelmp, 2017-04-26:
Ok thanks for the help, much appreciated mate, I will order the parts. Problem solved.

closed 2017-04-26 by elfhelmp