google/OctoPrint-LEDStripControl #5: Feature Request: UI button to run M150 command

xstrex, 2017-04-25:
So far the plugin is working great, now that I've got it figured out!

Would it be possible to add a button or two to the main UI, perhaps in the nav-bar, to run a M150 command, and turn the lights on, or off? This assumes the plugin is working, as designed. Ideally you could configure what the button did; like button A = "M150 R U B" & B = "M150 R255", etc.

xstrex, 2017-04-25:
Currently, in order to manually turn the lights on, off, or some color, you'd have to add the command to the start or stop gcode, which is fine. But let's say you wanted to turn the lights on to change filament, or do some bed leveling? Seems a bit much to open the UI, goto console, type "M150 R U B". A button would streamline this process. 

Also would be nice if the LEDs could be controlled via the API, and thus by other devices on the network!

precision, 2017-04-26:
You should be able to add buttons to the control screen for what you via the standard [Octoprint config.yaml.](http://docs.octoprint.org/en/master/configuration/config_yaml.html#controls)  The below snippet is untested, but I think it should work.

```
controls:
- children:
  - command: M150 R
    name: Red
    type: command
  - command: M150 U
    name: Green
    type: command
  - command: M150 B
    name: Blue
    type: command
  - command: M150 RUB
    name: White
    type: command
  - command: M150
    name: Off
    type: command
  layout: horizontal
  name: LEDs
```

As for the API, I'll add it as a possible future feature request.

xstrex, 2017-04-27:
Thanks for the feedback, the above "controls" block, worked perfectly in the config.yaml file! I look forward to the addition of this plugin to the API, what would be great!

closed 2017-04-27 by xstrex