google/OctoPrint-LEDStripControl #7: Noob question

suprdave04, 2017-05-23:
Have plugin installed but cannot seem to get the led strip to light up.  I have a maker select plus V2 printer, rgb from strip are connected to 19, 21, and 23 physical  gpio pins on my Pi 3. Power hooked up to pin 1 on the Pi.  Set those physical pin numbers in the plugin as described in the instructions.    I also tried 11, 13, and 15 on the pi as well, no go..

Also ran the usermod command described in the instructions, but one thing to note is that I needed to run the command with sudo in front of all of it.

I'm sure I'm missing something..

Any help would be greatly appreciated..

Thanks

precision, 2017-05-23:
Have you checked the Octoprint logs?  The plugin is pretty verbose about permissions issues.  ~/.octoprint/logs/octoprint.log  If you didn't, you'll need to restart octoprint/reboot after you execute the usermod for the group changes to take effect.

precision, 2017-06-16:
No response, please reopen if you're still having this issue.

ShagoY, 2018-04-02:
Same here... I was try everything (Fresh Octopi and also try on Raspbian+Octoprint)... Update everything, usermod and i check conf with physical Pin... Reboot a lot...  but my strip light don't change color.
Logs are fine : 
octoprint.plugins.LEDStripControl - DEBUG - M150 Detected: M150 R255
octoprint.plugins.LEDStripControl - DEBUG - match 1: r 2: 100.0

Btw i can change my color light by using pigpiod (but i need to switch off Octoprint).

So now, i have no more idea... I think i have trouble with my led strip RBG 5050 12V (B072J41Q2Z on amazon), what reference did you use ?

precision, 2018-04-02:
@ShagoY if you can change the colors w/ pigs then it should be supported fine.  Are you sure you've mapped the BCM pins to the pigs pins correctly?  For consistency the plugin takes BCM pin numbers.

ShagoY, 2018-04-02:
Yes, i use BCM 17/27/22, so physical PIN 11/13/15 on your GPIO configuration.
I also try BCM 23/24/25.. with no success.

I was follow this tuto : http://www.instructables.com/id/Easiest-RGB-LED-Strip-With-Raspberry-Pi/ (Give by https://www.thingiverse.com/thing:2803771/)

precision, 2018-04-02:
@ShagoY what happens if you try it w/o pigpio?  Should just be able to stop the daemon, change the plugin, restart octoprint and test.

ShagoY, 2018-04-02:
I was install pigpio(d) only for testing (i don't want run pigpiod on boot).

My config : 
  LEDStripControl:
    _config_version: 2
    b: 13
    g: 15
    on_startup: false
    r: 11
(So no use pigpiod)

More Octoprint Logs : 
octoprint.plugins.LEDStripControl - DEBUG - LEDStripControl on_settings_load()
octoprint.plugins.LEDStripControl - DEBUG - _register_leds()
octoprint.plugins.LEDStripControl - DEBUG - got pin(11)
octoprint.plugins.LEDStripControl - DEBUG - _setup_pin(11)
octoprint.plugins.LEDStripControl - DEBUG - got pin(15)
octoprint.plugins.LEDStripControl - DEBUG - _setup_pin(15)
octoprint.plugins.LEDStripControl - DEBUG - got pin(13)
octoprint.plugins.LEDStripControl - DEBUG - _setup_pin(13)
octoprint.plugins.LEDStripControl - DEBUG - got pin(0)
octoprint.plugins.LEDStripControl - DEBUG - _setup_pin(0)

Here my config : 
[image]

precision, 2018-04-03:
@ShagoY this pins do look correct according to the picture.  The logs you show look fine, no errors at all.  Is there anything logged if you send a "M150 R" to the terminal tab of Octoprint?

ShagoY, 2018-04-09:
Hi Precision :)
It's "work" now.. I think my last led strip are broken :(

Thank for all !

precision, 2018-04-09:
Good to hear!

closed 2018-04-09 by precision