google/OctoPrint-LEDStripControl #9: Will this work on common anode RGB LEDs? If not, what modifacations would need to be made to make it work?

MrBreadWater, 2017-07-17:
Title says it all. Thanks for your help!

precision, 2017-07-17:
Yes, that is what I use.

MrBreadWater, 2017-07-17:
How do I close this issue...?

closed 2017-07-18 by precision

MrBreadWater, 2017-10-11:
@precision Hey, sorry to bring this up again, but I'm having a bit of trouble making colors. Since it's common anode, everything is reversed. So 255 is actually 0, and vice versa. Is there a way to change this to be the opposite?

MrBreadWater, 2017-10-11:
@precision actually, scratch that, something else is going on. Setting Blue to 255 will make blue, setting red to 255 makes red, etc, but I can't mix colors unless they are perfectly equal. If they aren't it will just be whichever color has a higher value, but the led will just be dimmer than normal.

precision, 2017-10-11:
That is odd.  Few questions..
1. How are your LEDs/mosfets wired up to the Pi?
2. Are you using pigpiod?  
3. What Pi are you using? 
4. What do you see if you send 'M150 R200 G0 B200'?

MrBreadWater, 2017-10-12:
1. I have 2 wired up.
2. No.
3. Pi Zero W.
4. I see a light green. R255 U0 B255 gives just green.

For some unknown reason, color mixing started working. But I still have the original problem of 255 being off and 0 being on.

MrBreadWater, 2017-10-12:
@precision Ok, earlier I had read and blue swapped, and I had green set to the wrong pin altogether. That was why R255 would make red and b255 would make blue. Now R255 makes turquoise, and B255 makes yellow, and U255 makes pink/purple. 

It seems like the problem is now the inversion. Simple to fix, however, I don't understand your code (I'm a python amateur) otherwise I would fix it myself. All you need to do is take whatever value it should be, and subtract it from 255. Would you be so kind as to send me a copy of the python with that change?

precision, 2017-10-12:
I think all you should have to do to invert the values is to change line 136 of __init__.py

```
				try:
					v = float(match.group(2))
				except ValueError:
					# more than likely match.group(2) was unspecified
					v = 255.0
```

And just change it to
```
				try:
					v = 255.0 - float(match.group(2))
				except ValueError:
					# more than likely match.group(2) was unspecified
					v = 0.0
```

MrBreadWater, 2017-10-12:
Oh is that all? Well, in that case, that makes things much easier. Do you
know where the file is stored? I can't find it.

precision, 2017-10-12:
Well assuming you're using Octopi it should be in 

> /home/pi/oprint/lib/python2.7/site-packages/octoprint_LEDStripControl/

Of not, basically the same path but where ever you put your virtualenv base.

MrBreadWater, 2017-10-12:
Alright, I've changed it and it pretty much works. I just have to specify
all the values, otherwise the led defaults to white. But other than that it
works flawlessly.

MrBreadWater, 2017-10-12:
Thanks for all your help!

precision, 2017-10-13:
Glad to hear!