google/OctoPrint-LEDStripControl #10: Invalid gcode command

charicner, 2017-08-11:
when I try to use this I get a response from the pi that it is an invalid gcode command. I am not sure how to correct this. I have verified the GPIO pins, and made sure they are related to the correct ones in the software, but it looks like the pi is not intercepting the commands.

precision, 2017-08-11:
Have you enabled the plugin under the Octoprint settings?  By default it is installed, but disabled.

antonioreyna, 2018-01-25:
i have the same issue, it even stops the print completly when sending the code

precision, 2018-01-25:
Can you attach a screenshot of the configuration screen for the LEDStrip Plugin??

antonioreyna, 2018-01-26:
[image]

antonioreyna, 2018-01-26:
it turns and change the colours correctly, but i see that message on console invalid gcode command, my printer is an M3D Micro, is it possible that the printer is the one causing the problem? or it doesn't matter that it does not come with LEDs to control and it must accept the gcode?

precision, 2018-01-26:
Aha, I guess I misunderstood the plugin API docs the first time around.  I'll push a fix tomorrow morning.

precision, 2018-01-26:
If you want to try it out yourself, you can manually force octoprint to install from my [dev branch](https://github.com:precision/OctoPrint-LEDStripControl.git)

antonioreyna, 2018-01-26:
im in the middle of a print if tomorrow is not released yet ill test the dev, thanks for the support! :D

precision, 2018-01-26:
I just pushed 0.2.1 which should be properly swallowing the M150.

antonioreyna, 2018-01-26:
now it works from terminal, can i add this command directly to the gcode of the object to print? because if i dont add it the print works great, but when i add the command to the gcode file after some lines i get this: 

Send: N22 M135 T0*80
Recv: T:219.6360
Recv: ok Unknown G-code command

what do you think is happening?

antonioreyna, 2018-01-26:
could it be ignoring your new hook when it comes from the object gcode?

precision, 2018-01-26:
Are you printing gcode files from SD?

antonioreyna, 2018-01-26:
yes well i upload them to the server its a octopi, im about to do a new test removing that command M135 it is being added by simplify3d on my cura files that code is not there, so maybe thats the issue

precision, 2018-01-26:
The plugin doesn't touch M135, just M150.   So I'm at a bit of a loss here.

antonioreyna, 2018-01-26:
yes sorry for that!! this was a different problem  now the plugin with my prints works great that other command i just told you was a separate non related issue after the 150 as i had this on my gcode...

```
M150 R255 U255 B100
G92 E0
G1 E-5.0000 F1200
G1 Z0.200 F1002
; process Process1
; layer 1, Z = 0.200
M135 T0
```

it was first crashing on the M150 then you fixed it and i thought for a moment the problem continued but no now i removed the 135 from my code and now its done 

thanks for the support and the time you took to fix the plugin :D

precision, 2018-01-26:
Glad to hear its working correctly now!

closed 2018-01-26 by precision