google/OctoPrint-LEDStripControl #14: Not showin up in Plugins tab

J0nny12, 2018-02-25:
Hi, i have following problem:
after successfully installing your plugin it does not show up in the plug-in tab to configure it.
Error message "A plugin was installed successfully, however it was impossible to detect which one. Please Restart OctoPrint to make sure everything will be registered properly" is displayed.
After the restart, it wont show up in plugin manager not the config tab.
please help :(

best regards
Jonny

trev502000, 2018-02-27:
i also have the exact same problem

precision, 2018-02-28:
What version of Octoprint are you running?  Was there anything in the octoprint logfile?  ~/.octoprint/logs/octoprint.log

trev502000, 2018-02-28:
im running octoprint 1.3.6 
the log is attached
[octoprint.log](https://github.com/google/OctoPrint-LEDStripControl/files/1766733/octoprint.log)

J0nny12, 2018-02-28:
When i first issued this, i was on 1.3.4
now at 1.3.6, no difference.
[octoprint.log](https://github.com/google/OctoPrint-LEDStripControl/files/1767377/octoprint.log)

jean62, 2018-02-28:
I also have the exact same problem
octoprint 1.3.6
[octoprint.log](https://github.com/google/OctoPrint-LEDStripControl/files/1767600/octoprint.log)

precision, 2018-02-28:
Are these all raspberry pis?  It appears they don't find the pigpiod python package, which is included in the later versions of Raspbian/Octopi.  Maybe old versions of OS?

jean62, 2018-02-28:
[image]

how to update octoPi

J0nny12, 2018-03-01:
@precision pigpio library is installed and working fine. I use some plugin to read out Gcode to execute gpio states with pigpio...basically what i wanted to do with this plugin, but worse :)

precision, 2018-03-01:
@J0nny12 can you try this.. as the user pi and paste the output?

```
source ~/oprint/bin/activate
pip search pigpio
```

J0nny12, 2018-03-01:
@precision 
[image]

precision, 2018-03-01:
Weird, thanks.  I'm doing a little more investigation on my side.  I was just able to reproduce this on one of my Pis.

J0nny12, 2018-03-01:
@precision really appreciate your work!
Looking forward to not using my sh..... solution anymore :-)

precision, 2018-03-01:
Sigh.  It looks like they changed something in the python pigpio library.  For now you can work around this with until I can figure out what they did.

```
source ~/oprint/bin/activate
pip install pigpio==1.3.5
```

Then reinstall the plugin via the plugin manager.

precision, 2018-03-01:
For now, I bumped the plugin version and pinned pigpio to 1.35 which works here for me. 

0ad7fc5b5a0dbdd21e5b746031c1b564573cdf7d #

J0nny12, 2018-03-01:
Thanks, works again!

precision, 2018-03-01:
After a little more investigating, it seems the 1.40 package from pypi doesn't contain any of the actual python code.  The fixed version (0.2.3) with the pinned pigpio package I pushed should solve this for the time being.

precision, 2018-03-07:
Looks like they fixed the package in 1.40.post1.  I'm going to stick with 1.35 pinned for now, but will update it later if the need arises.

closed 2018-03-07 by precision