google/OctoPrint-LEDStripControl #17: KeyError: 'w'

gulliverrr, 2018-03-24:
Hello
I can control the LED via python manually but not if I send an M150 W command for my only-white LEDs. I see these in /home/pi/.octoprint/logs/octoprint.log : 

```
2018-03-24 00:02:41,284 - octoprint.util.comm - ERROR - Error while processing hook LEDStripControl for phase queuing and command M150:
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/util/comm.py", line 2419, in _process_command_phase
    hook_results = hook(self, phase, command, command_type, gcode, subcode=subcode)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_LEDStripControl/__init__.py", line 147, in HandleM150
    if self._leds[l]:
KeyError: 'w'
```

Any clues?

gulliverrr, 2018-03-24:
I noticed this on startup too
```
2018-03-24 00:35:56,515 - octoprint.server - ERROR - Error while trying to migrate settings for plugin LEDStripControl, ignoring it
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/server/__init__.py", line 395, in run
    settings_plugin_config_migration_and_cleanup(implementation._identifier, implementation)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/server/__init__.py", line 386, in settings_plugin_config_migration_and_cleanup
    implementation.on_settings_initialized()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_LEDStripControl/__init__.py", line 170, in on_settings_initialized
    self._register_leds()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_LEDStripControl/__init__.py", line 114, in _register_leds
    self._leds[i] = self._setup_pin(pin)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_LEDStripControl/__init__.py", line 95, in _setup_pin
    p.start(startup)
ValueError: dutycycle must have a value from 0.0 to 100.0
```

precision, 2018-03-29:
Sorry, I was off at MRRF over the weekend and didn't get time to look at this.  The first key error is harmless and will go away if you have RGBW strips, but to be safe I added a try/catch loop.

closed 2018-03-29 by precision

gulliverrr, 2018-03-29:
When you say  RGBW strips do you mean addressable ones or plain 5-contact ones (+12V, R, G, B, W)?
I only installed plain white ones expecting the pin 12 (entered as 12 in the plugin's page) that is BCM18 (which is also a PWM pin too!) to go HIGH or LOW when an M150 W would arrive but nothing happens. Any hints?

precision, 2018-03-29:
I meant the RGBW nonaddressable ones.  I just bumped the version of the plugin to 0.3.3 which will pull in the fix I post earlier.  If that version doesn't work I'll need some more logging information.

gulliverrr, 2018-04-01:
How do I push the 0.3.3. Even forcing check from integrated plugin manager does not find anything newer than 0.3.2

precision, 2018-04-01:
It should just update, but you can try just removing it and reinstalling.  That should grab the master.zip from the tree.  If not manually install a plugin via the manage with the following URL: https://github.com/google/OctoPrint-LEDStripControl/archive/master.zip

gulliverrr, 2018-04-02:
Yeap, disabling and uninstalling and reinstalling fixed it and pulled v 0.3.6 which also fixes my issue and works perfectly now! Thank you so much!

commit 9891cc2d (2018-03-29): fixes #17