google/OctoPrint-LEDStripControl #18: Plugin does not save\function

bi0hazard1980uk, 2018-03-24:
After installing the latest version with White control I am unable to get the plugin to work. It will only save the GPIO configuration once and then requires a reboot. But I cannot get it to control the LED's yett the old version works fine.

gulliverrr, 2018-03-28:
Which version is working for you as the latest does not work for me?

precision, 2018-03-29:
Hrm, it sounds maybe a version itis type thing in ~/.octoprint/config.yaml.  If you know your GPIO pins try removing everything from there and let it set up fresh values.

precision, 2018-03-29:
I need more info to diagnose this, can you grab the octoprint.log for me?

ShagoY, 2018-03-30:
2018-03-31 01:58:46,484 - octoprint.plugins.LEDStripControl - DEBUG - LEDStripControl on_settings_save()
2018-03-31 01:58:46,486 - octoprint.plugins.LEDStripControl - DEBUG - _unregister_leds()
2018-03-31 01:58:46,491 - octoprint.plugins.LEDStripControl - DEBUG - _register_leds()
2018-03-31 01:58:46,493 - octoprint.plugins.LEDStripControl - DEBUG - got pin(13)
2018-03-31 01:58:46,494 - octoprint.plugins.LEDStripControl - DEBUG - _setup_pin(13)
2018-03-31 01:58:46,497 - octoprint.server.api.settings - ERROR - Could not save settings for plugin LED Strip Control (0.3.3)
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/api/settings.py", line 485, in _saveSettings
    plugin.on_settings_save(data["plugins"][plugin_id])
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_LEDStripControl/__init__.py", line 182, in on_settings_save
    self._register_leds()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_LEDStripControl/__init__.py", line 117, in _register_leds
    self._leds[i] = self._setup_pin(pin)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_LEDStripControl/__init__.py", line 95, in _setup_pin
    p.start(startup)
ValueError: dutycycle must have a value from 0.0 to 100.0

precision, 2018-03-31:
Thank you, this helps me immensly. I'll investigate when I get time to
night.

-U

precision, 2018-03-31:
I just commited a quick fix that I think will work.  I won't be back at my printer witht the LEDs on it to test this out by hand.  Sorry about the flakiness of this, I should have reverted the initial patch that caused this instead of trying to fix it on the fly.

ShagoY, 2018-03-31:
OctoPrint 1.3.6
octoprint.util.pip - INFO - Version of pip is 9.0.1
LED Strip Control (0.3.4) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_LEDStripControl

no change : 

2018-04-01 - py.warnings - WARNING - /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_LEDStripControl/__init__.py:109: RuntimeWarning: No channels have been set up yet - nothing to clean up!  Try cleaning up at the end of your program instead!
  GPIO.cleanup()

2018-04-01 - octoprint.server.api.settings - ERROR - Could not save settings for plugin LED Strip Control (0.3.4)
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/server/api/settings.py", line 462, in _saveSettings
    plugin.on_settings_save(data["plugins"][plugin_id])
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_LEDStripControl/__init__.py", line 182, in on_settings_save
    self._register_leds()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_LEDStripControl/__init__.py", line 117, in _register_leds
    self._leds[i] = self._setup_pin(pin)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_LEDStripControl/__init__.py", line 94, in _setup_pin
    p = GPIO.PWM(pin, float(startup))
ValueError: frequency must be greater than 0.0

precision, 2018-03-31:
@ShagoY if you could, please try 0.3.5 which I just pushed in that commit.

ShagoY, 2018-04-01:
I delete entry on my config/yaml after reinstalling but when i change value, i got the same error : 

LED Strip Control (0.3.5) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_LEDStripControl

py.warnings - WARNING - /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_LEDStripControl/__init__.py:109: RuntimeWarning: No channels have been set up yet - nothing to clean up!  Try cleaning up at the end of your program instead!
  GPIO.cleanup()

octoprint.server.api.settings - ERROR - Could not save settings for plugin LED Strip Control (0.3.5)
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/server/api/settings.py", line 462, in _saveSettings
    plugin.on_settings_save(data["plugins"][plugin_id])
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_LEDStripControl/__init__.py", line 182, in on_settings_save
    self._register_leds()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_LEDStripControl/__init__.py", line 117, in _register_leds
    self._leds[i] = self._setup_pin(pin)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_LEDStripControl/__init__.py", line 95, in _setup_pin
    p.start(float(startup))
ValueError: dutycycle must have a value from 0.0 to 100.0

precision, 2018-04-01:
Thank you for testing for me.  I'm going to setup a RPi.GPIO system tomorrow and get this fixed once and for all.

precision, 2018-04-01:
I just uploaded a new version.  I didn't realize that Octoprint wouldn't transparently migrate configurations to new settings.   I tested this version with old (RGB) configurations under both pigpiod and rpi.gpio and they upgraded and worked fine.  It should install as 0.3.6.

ShagoY, 2018-04-02:
No more error on log. thx

precision, 2018-04-02:
Great to hear, thanks!

commit c8101d74 (2018-03-31): Trying fix #18, this is what I get for a trying to do a quick patch
while at the airport.
closed 2018-04-02 by precision