google/OctoPrint-LEDStripControl #21: Cannot trigger with M150 command (Prusa I3 Mk2s)

mikedmor, 2018-05-08:
I setup everything and got my pins all set, ran some test via ssh with pigpiod to make sure everything works (which it is), and updated my Prusa I3 firmware to enable BLINKM. However when i open octoprint and input M150 R255 there is no reponse, and the led's do not turn on. I also attempted to use the Pigpiod settings and still was unable to get my leds to turn on. I am unsure if this issue has to do with my printer or if it is related to the plugin.

precision, 2018-05-08:
What version of Octoprint and Raspbian are you using?

mikedmor, 2018-05-08:
Octoprint Version 1.3.8
Raspbian Version 8 (jessie)

precision, 2018-05-08:
Ok, I think I saw something similar when I upgraded to Jessie, but I'm not 100% sure.   Can you try the following?

I'm assuming this is a OctoPi like setup, where there is a virtualenv in ~/oprint

```
source oprint/bin/activate
pip install pigpio --upgrade
```

Then restart OctoPrint and try the M150 again.  If that fixes it, I'll push a new version of the plugin that grabs the latest version of the pigpio python module.  I had it pinned @ 1.35 because there was a broken version in the wild for a while.

mikedmor, 2018-05-08:
Yes this is an OctoPi setup.

Still no luck though. I tried with both the pigpiod setting enabled and disabled again.

precision, 2018-05-08:
Ok.  So it doesn't work with or without pigpiod.  What pins do you have the plugin configured to use and what pins on your RPi are your LEDs wired to?

mikedmor, 2018-05-08:
Im using pins 17 for red, 22 for green, and 24 for blue. I followed this tutorial to get all the wiring setup: https://dordnung.de/raspberrypi-ledstrip/#!%2Fws2812

mikedmor, 2018-05-08:
Well I did find out i was configuring the tool wrong. Was using GPO numbers instead of physical pin numbers, However after updating to the physical pins it still does not seem to work. The correct pins i have setup is 11 for Red, 15 for Green, and 18 for Blue.

mikedmor, 2018-05-08:
They just turn on! So i guess it took a moment. Thank you for the help!

precision, 2018-05-08:
Great to hear!  Glad it was just a configuration issue.

closed 2018-05-08 by mikedmor