google/OctoPrint-LEDStripControl #22: M150 Codes Ignored By Terminal

jmwright, 2018-08-09:
I'm using Octoprint 1.3.9 on OctoPi 0.15.1. I don't have the box checked for `Use pigpiod` and pigpio is not installed. The pi user is in the gpio group and I'm using the physical pin numbering scheme (13 for red). When I enter M150 codes in the terminal I never get a response back, and the inputs never change. I don't see any errors in the log related to this. I did add the buttons suggested in #20 but they don't have any effect either. Any ideas on what to try next?

precision, 2018-08-10:
First thing, double check the pins via https://pinout.xyz/

Second thing, try enabling debugging output to the log file.  Add the following to ~/.octoprint/logging.yaml (create it if it doesn't exist)  You'll need to reboot or restart Octoprint for it to take effect.  Then check the octoprint.log to see if there is anything useful being logged.
```
loggers:
  octoprint.plugins.LEDStripControl:
    level: DEBUG
```

jmwright, 2018-08-10:
It works now.

Below is the only new thing I saw in the log. I had added the buttons to the Control tab and pressed the "Red" button once, and the "Turn Off" button once.
```
2018-08-10 17:24:16,720 - octoprint.plugins.LEDStripControl - DEBUG - M150 Detected: M150 R
2018-08-10 17:24:16,721 - octoprint.plugins.LEDStripControl - DEBUG - match 1: r 2: 100.0
2018-08-10 17:25:57,622 - octoprint.plugins.LEDStripControl - DEBUG - M150 Detected: M150
```

I was originally sending `M150 R255` from the terminal because that's what I saw in the readme, but noticed the "Red" button was only sending `M150 R`. Once I started sending that, it started working. Thanks.

closed 2018-08-10 by jmwright