google/OctoPrint-LEDStripControl #24: Cabling + Compatibility Ender 3

pika28, 2019-01-15:
Hello,

- Does it work with the Ender 3 in the original configuration?-
- Where should I enter this command line : "usermod -a -G gpio pi" ?

My biggest problem is the wiring hardware part of the led. How to wire a RGB 5050 LED strip? I have 4 solder contacts: 12V, RED, GREEN and BLUE.

Bests Regards

precision, 2019-01-15:
LEDStripControl works via LEDs connected to a Raspberry Pi / SBC's GPIO pins.  It has nothing to do with the printer itself.

closed 2019-01-15 by precision

pika28, 2019-01-15:
No need for a controller? I directly cable 12V, R, G and B on the pins of the PI? There is no 12V in the PI ...

pika28, 2019-01-15:
A great tutorial indeed! There is no module ready for use? I do not have the hardware to create a print circuit.

precision, 2019-01-15:
I use a board a friend designed.  You could have OSHPark make the PCB and solder up the components.

https://github.com/thatch/step_down_led_board/tree/v1.x

pika28, 2019-01-16:
I am not used to this site. I find a photo of the PCB but can we find the list of components? How do we go from the website to the online order? Sorry for all these questions

----- Mail original -----

https://github.com/thatch/step_down_led_board/tree/v1.x

precision, 2019-01-16:
I've asked my friend for a bill of materials.  If I remember correct we just used random components we found lying around in the shop. 

But if I were you, I'd just breadboard/perfboard up the schematic in the URL I linked previously.   It is super simple, just some mosfets, a barrel connector, and some wires.