google/OctoPrint-LEDStripControl #26: Orange Pi Support

Bensmall, 2019-02-16:
Does this work with Armbian on orange pi zero ? have installed the plugin and got it to output via ssh and WiringOP But not able to get a response from the octoprint interface

precision, 2019-02-17:
I did successfully use it on a OrangePI a year or so ago with WiringPi.  However the flickering on the LEDs was too much for me handle.

closed 2019-02-17 by precision

Bensmall, 2019-02-17:
being new at this, you wouldn't happen to be able to tell me how you got it to work? haven't had any luck so far 

Ben

Bensmall, 2019-02-17:
all i get in the log is 
octoprint.util.comm - ERROR - Error while processing hook LEDStripControl for phase queuing and command M150:
Traceback (most recent call last):
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/util/comm.py", line 3085, in _process_command_phase
    hook_results = hook(self, phase, command, command_type, gcode, subcode=subcode, tags=tags)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_LEDStripControl/__init__.py", line 149, in HandleM150
    if self._leds[l]:
KeyError: 'w'

precision, 2019-02-19:
That looks like an old bug #17 ?