google/OctoPrint-LEDStripControl #27: Unable To Save Config Without Reboot Then Freeze

thomasdlap, 2019-02-24:
Hey,

Hoping to get some help, I just recently installed LedStripControl. Unfortunately I haven't been able to try it out due to flakiness.

installed from Octoprint plugin manager 

Once installed, ill try to set the following pins in the config ui

8, 10, 12, 15

upon clicking save, the save spinner will just sit their spinning(waited up to 10 minutes), after forcing a reboot, I am greeted with a white screen when trying to boot into touchui.
[image]
[image]

I then went into the config.yaml and found the following config

 ```
 LEDStripControl:
    _config_version: 2
    b: 12
    g: 10
    on_startup: false
    r: 8
    w: 15
```

I then removed the config and performed a reboot which allowed Octoprint to boot normally but puts me right back to where I started. 

Any help would be greatly appreciated

Raspberry Pi 3 B+, used Raspyberry Pi 7" touchscreen with TouchUI
Printer: Prusa i3 MK$
OctoPrint version : 1.3.10
OctoPi version    : 0.15.1

precision, 2019-02-25:
I have no experience with TouchUI so I'm a shooting in the dark here.  Are there anything in ~/.octoprint/logs/octoprint.log?

The plugin itself doesn't add anything to the UI so I don't think it could be something there.  But can you get to a javascript console in TouchUI?

thomasdlap, 2019-02-25:
Thanks for the quick response @precision 

Heres a dump of the .octoprint/logs/octoprint.log, I've also included a screenshot as to what im currently seeing when trying to log into Octoprint ui. Note that removing the LEDStripControl config in 
~/.octoprint/config.yaml will fix the problem

[image]

```

2019-02-25 22:26:45,251 - octoprint.startup - INFO - ******************************************************************************
2019-02-25 22:26:45,253 - octoprint.startup - INFO - Starting OctoPrint 1.3.10
2019-02-25 22:26:45,254 - octoprint.startup - INFO - ******************************************************************************
2019-02-25 22:26:45,328 - octoprint.startup - INFO - We don't appear to be online, not fetching plugin blacklist
2019-02-25 22:26:45,405 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin pack$
2019-02-25 22:26:48,062 - octoprint.plugin.core - INFO - Found 17 plugin(s) providing 17 mixin implementations, 21 hook handlers
2019-02-25 22:26:48,191 - octoprint.server.heartbeat - INFO - Starting server heartbeat, 900.0s interval
2019-02-25 22:26:48,314 - octoprint.server - INFO - Intermediary server started
2019-02-25 22:26:48,314 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin pack$
2019-02-25 22:26:48,905 - octoprint.plugin.core - INFO - Found 17 plugin(s) providing 17 mixin implementations, 21 hook handlers
2019-02-25 22:26:48,948 - octoprint.filemanager.storage - INFO - Initializing the file metadata for /home/pi/.octoprint/uploads...
2019-02-25 22:26:49,610 - octoprint.filemanager.storage - INFO - ... file metadata for /home/pi/.octoprint/uploads initialized successfully.
2019-02-25 22:26:49,693 - octoprint.plugins.softwareupdate - INFO - Loaded version cache from disk
2019-02-25 22:26:53,455 - octoprint.util.pip - INFO - Using "/home/pi/oprint/bin/python2 -m pip" as command to invoke pip
2019-02-25 22:26:55,521 - octoprint.util.pip - INFO - Version of pip is 9.0.3
2019-02-25 22:26:55,523 - octoprint.util.pip - INFO - pip installs to /home/pi/oprint/lib/python2.7/site-packages (writable -> yes), --user flag needed -> no, virtual env -> yes
2019-02-25 22:26:55,524 - octoprint.util.pip - INFO - ==> pip ok -> yes
2019-02-25 22:26:55,529 - octoprint.plugin.core - INFO - Initialized 17 plugin implementation(s)
```

thomasdlap, 2019-02-26:
After doing some more debugging, not really sure what im doing but having fun :)

The below line self._pigpiod = pigpio.pi() is where i hang up as far as i can tell, If i comment out this line the debugger is able to proceed 

```
 if self._pigpiod is None:
                                self._pigpiod = pigpio.pi()
```
[image]

thomasdlap, 2019-02-26:
Looking into the docs of http://abyz.me.uk/rpi/pigpio/python.html#pigpio.pi

if I specify the ip address and port, it seems to resolve the issue 

```
self._pigpiod = pigpio.pi('192.168.0.13', '8080')
```

precision, 2019-02-26:
Huh that is strange.. By default it should connect to 'localhost'.  Is it possible that you don't have localhost in /etc/hosts?

Jrami608, 2019-03-23:
Any updates on this? I am having the same issue and can't figure it out.

precision, 2019-03-25:
Can one of you paste in the output of
```
host localhost && cat /etc/hosts
```

And assuming this is octopi (if not you'll need to substitute the path to your virtualenv)

```
source ~/oprint/bin/activate && pip list
```

closed 2019-07-15 by precision