google/OctoPrint-LEDStripControl #29: After updating to newest octoprint version - only

GunnarFrei, 2019-05-23:
I've just updated my octoprint to version 1.3.11
and it only allows me to boot in security mode.
When i deactivate LED Strip Control  0.3.6 in Plugin Manager octoprint boots normal again. 
Bug in v0.3.6?

preloi, 2019-05-26:
I have exact the same Problem here.
Can you provide a update?

precision, 2019-05-28:
I just updated my two test instances and both seem to work fine.  Is there by chance anything in the logs?  Have any of you tried removing/reinstalling the plugin?

ViPeR5000, 2019-05-28:
yes try reinstalling and dont work :(
only removing fix the octoprint

OctoPrint version : 1.3.11
OctoPi version    : 0.16.0

log
019-05-28 21:13:48,192 - octoprint.plugins.LEDStripControl - ERROR - Unable to communicate with PiGPIOd
2019-05-28 21:13:48,599 - octoprint.cli.server - ERROR - Uncaught exception
Traceback (most recent call last):
  File "/home/pi/oprint/bin/octoprint", line 11, in <module>
    load_entry_point('OctoPrint==1.3.11', 'console_scripts', 'octoprint')()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/__init__.py", line 609, in main
    octo(args=args, prog_name="octoprint", auto_envvar_prefix="OCTOPRINT")
  File "/home/pi/oprint/local/lib/python2.7/site-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/pi/oprint/local/lib/python2.7/site-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/click/decorators.py", line 17, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/cli/server.py", line 206, in serve_command
    ignore_blacklist)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/cli/server.py", line 118, in run_server
    octoprint_server.run()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/__init__.py", line 478, in run
    extra=dict(plugin=plugin._identifier))

precision, 2019-05-28:
It looks like pigpiod isn't responding/running.  I just want to make sure that this isn't something strange like #27 

Do you have pigpiod installed and running?
Do you have localhost in /etc/hosts?

ViPeR5000, 2019-05-28:
/etc/hosts

127.0.0.1       localhost
::1             localhost ip6-localhost ip6-loopback
ff02::1         ip6-allnodes
ff02::2         ip6-allrouters

127.0.1.1       octopi

and pigiod working

precision, 2019-05-29:
@ViPeR5000 Can you tell me what version of pigpio you have installed?

```
source ~/oprint/bin/activate # or wherever your octoprint virtualenv is
pip list | grep pigpio
```

ViPeR5000, 2019-05-29:
> @ViPeR5000 Can you tell me what version of pigpio you have installed?
> 
> ```
> source ~/oprint/bin/activate # or wherever your octoprint virtualenv is
> pip list | grep pigpio
> ```
(oprint) pi@octopi:~ $ pip list | grep pigpio
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7.
pigpio                       1.35

precision, 2019-05-29:
Hrm, ok that is the same version I have on my test instances as well.  Can you paste the plugins -> LEDStripControl section from your ~/.octoprint/config.yaml? 

Thanks alot for helping track this down.

ViPeR5000, 2019-06-03:
LEDStripControl:
    _config_version: 2
    b: 24
    g: 22
    pigpiod: true
    r: 17

precision, 2019-06-03:
I think I might have found the issue.  It seems that the default service file for pigpiod on the latest OctoPi image doesn't set pigpiod to listen on localhost.  I must have fixed this myself on my test instances without thinking.

Try this and see if it fixes it for you
```
$ sudo cat << EOF | sudo tee /etc/systemd/system/pigpiod.service
[Unit]
Description=Daemon required to control GPIO pins via pigpio
[Service]
ExecStart=/usr/bin/pigpiod -l -n 127.0.0.1
ExecStop=/bin/systemctl kill pigpiod
Type=forking
[Install]
WantedBy=multi-user.target
EOF

$ sudo systemctl daemon-reload
$ sudo systemctl restart pigpiod
```
After that try restart OctoPrint and see if that fixes it.

In case you're interested here's the diff from the package installed service file and mine.
```
$ diff -u /lib/systemd/system/pigpiod.service /etc/systemd/system/pigpiod.service 
--- /lib/systemd/system/pigpiod.service	2017-01-12 07:40:41.000000000 -0800
+++ /etc/systemd/system/pigpiod.service	2018-11-22 10:05:37.967742093 -0800
@@ -1,7 +1,7 @@
 [Unit]
 Description=Daemon required to control GPIO pins via pigpio
 [Service]
-ExecStart=/usr/bin/pigpiod -l
+ExecStart=/usr/bin/pigpiod -l -n 127.0.0.1
 ExecStop=/bin/systemctl kill pigpiod
 Type=forking
 [Install]
````

emptywill, 2019-07-10:
Hi, i'm getting the same problem... after digging for a bit, here is what what i found in the logfile that seems to be the problem... Anything you guys can think of?

2019-07-10 01:31:42,026 - octoprint.server.api.settings - ERROR - Could not save settings for plugin LED Strip Control (0.3.6)
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/api/settings.py", line 550, in _saveSettings
    plugin.on_settings_save(data["plugins"][plugin_id])
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_LEDStripControl/__init__.py", line 179, in on_settings_save
    self._register_leds()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_LEDStripControl/__init__.py", line 116, in _register_leds
    self._leds[i] = self._setup_pin(pin)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_LEDStripControl/__init__.py", line 94, in _setup_pin
    GPIO.setup(pin, GPIO.OUT)
ValueError: The channel sent is invalid on a Raspberry Pi

emptywill, 2019-07-10:
Ok i found out what was my problem and i think Viper5000 has the same problem. I used the GPIO pins numbers instead of the physical number...
Edited ~/.octoprint/config.yaml
FROM:
LEDStripControl:
_config_version: 2
b: 24
g: 22
pigpiod: false
r: 17

TO:
LEDStripControl:
_config_version: 2
b: 18
g: 15
pigpiod: true
r: 11

Rebooted and worked.

closed 2019-07-15 by precision