google/OctoPrint-LEDStripControl #30: After install, Octoprint can't start up

tvixen, 2019-07-21:
After I installed and configure it to 14, 0, 0 (only had one led strip going). 
And restarted Pi. Octoprint can't start up correctly.
Says "The OctoPrint server is currently not running"

After start up in safe mode, and disabling your plugin, it will start again.
I then deleted the plugin, and restartede octoprint again. 
Then installed it agin via the plugin manager, and rebooted. 
Same happend again .....

Any idea og what is going wrong ?

Br Tvixen

precision, 2019-07-21:
Can you double check this isn't #29

tvixen, 2019-07-21:
Well sure its the same, but how to correct this when it wont start up. And deleting and install again is not helping?

precision, 2019-07-21:
Quickest way is to edit ~/.octoprint/config.yaml and remove/fix up the LEDStripControl section.

tvixen, 2019-07-21:
Can't it be fixed in your plugin ?
So we can install a new version ?

tvixen, 2019-07-21:
Great.... It fucked up my whole octoprint server. And I have to start all over again, with a new image install. 
6 hours waitsed, because of your little programming error.

closed 2019-08-05 by precision