google/OctoPrint-LEDStripControl #33: Not working cositently

GaWr26, 2020-02-02:
Hey, thanks for the great plugin.

I do have an issue though. I use octolapse and have configured two scripts for before and after a picture is taken. In these scripts, I set the LEDs:

e.g. ledShoot.sh
```
#!/bin/bash

pigs p 17 255
pigs p 27 100
pigs p 22 100

exit 0
```

I think somehow this interferes with the plugin. Sometimes, the M150 command works but stops working after using my custom scripts. Is there a rights issue or something I'm missing?
What is the recomended way of starting pigiod?

Cheers

GaWr26, 2020-02-04:
Maybe this helps:

`2020-02-04 12:03:53,140 - octoprint.util.comm - ERROR - Error while processing hook LEDStripControl for phase queuing and command M150 R:
Traceback (most recent call last):
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/util/comm.py", line 3292, in _process_command_phase
    hook_results = hook(self, phase, command, command_type, gcode, subcode=subcode, tags=tags)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_LEDStripControl/__init__.py", line 150, in HandleM150
    self._leds[l].ChangeDutyCycle(dutycycles[l])
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_LEDStripControl/__init__.py", line 65, in ChangeDutyCycle
    self.start(dutycycle)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_LEDStripControl/__init__.py", line 54, in start
    self._pigpiod.set_PWM_range(self._pin, 100) # emulate RPi.GPIO
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/pigpio.py", line 1441, in set_PWM_range
    return _u2i(_pigpio_command(self.sl, _PI_CMD_PRS, user_gpio, range_))
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/pigpio.py", line 978, in _pigpio_command
    dummy, res = struct.unpack('12sI', sl.s.recv(16))
error: unpack requires a string argument of length 16
2020-02-04 12:03:54,063 - octoprint.plugins.LEDStripControl - DEBUG - M150 Detected: M150 R
2020-02-04 12:03:54,064 - octoprint.plugins.LEDStripControl - DEBUG - match 1: r 2: 100.0
2020-02-04 12:03:54,065 - octoprint.plugins.LEDStripControl - DEBUG - PiGPIOpin: ChangeDutyCycle() pin: 17
2020-02-04 12:03:54,066 - octoprint.plugins.LEDStripControl - DEBUG - PiGPIOpin: start() pin: 17`

precision, 2020-02-04:
For starting pigpiod, there should be an init script/service file included with the package `pigpiod` on Raspbian.

```
pi@raspberrypi:~ $ dpkg -L pigpiod | grep service
/lib/systemd/system/pigpiod.service
```

It looks like internally the pigpio library isn't returning the right data structure and it is tossing an exception.  I've never encountered that before.  What version of Raspbian and pigpiod is installed?

GaWr26, 2020-02-04:
Thanks for getting back @precision 

Here's my versions:
```
cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

```
pigpiod -v
71
```

precision, 2020-02-04:
Hrm. I'm kind of at a loss here.. this kind of just works on my Pi. What version of the pigpio python library do you have installed?

```
$ pigpiod -v
71
$ . ~/oprint/bin/activate && pip list | grep pigpio
pigpio                    1.35       
```

I'm connected, I can set the colors via M150 to whatever.  Then from the console I fire off a `$ pigs p 17 100` the blue (in my case) led goes to 100%.  If I then fire another M150 via the plugin it can still change the LEDs just fine.

GaWr26, 2020-02-04:
That is really strange. I also noticed if I enable the plugin and send a M150 command(which has no effect) the LEDs start to flicker.

`. ~/oprint/bin/activate && pip list | grep pigpio
-bash: /home/pi/oprint/bin/activate: No such file or directory
`
Maybe I should mention I installed octoprint manually following this guide: https://community.octoprint.org/t/setting-up-octoprint-on-a-raspberry-pi-running-raspbian/2337

precision, 2020-02-04:
Is your pigpiod listening on a tcp port or a socket?  I think the default behavior is to listen on a unix domain socket which can cause problems (permissions, apparmor, etc)  I always set mine to listen on a tcp port.

```
$ cat /etc/systemd/system/pigpiod.service 
[Unit]
Description=Daemon required to control GPIO pins via pigpio
[Service]
ExecStart=/usr/bin/pigpiod -l -n 127.0.0.1
ExecStop=/bin/systemctl kill pigpiod
Type=forking
[Install]
WantedBy=multi-user.target
```

GaWr26, 2020-02-04:
That did the trick!
`-n 127.0.0.1`

Unfortunately the scripts still interfere somehow and lights flicker with my scripts. 
I added the the controls snipped where I now have the buttons under the controls tab in octoprint. If I set all colors off via pigs
```
pigs p 17 0
pigs p 27 0
pigs p 22 0
```
and then use the buttons I have beautiful stable light. 
If I then switch lights with pigs:
```
pigs p 17 255
pigs p 27 100
pigs p 22 100
```
It flickers. Turning these off and using the controls in octoprint again works fine.

Any ideas?

Thanks a bunch for all your support!

precision, 2020-02-13:
Sorry for the delay, I wasn't able to notice any flickering on my setup. Possibly a quality difference on our LED strips?