google/OctoPrint-LEDStripControl #35: Plug-in enables two lights per one M150 command

professor305, 2020-07-18:
When entering the "M150" command 2 lights of the RGB lights come on. For example,

- the command "M150 R" turns green lights on as well as the requested "R" red lights, 

- the command "M150 G" turns blue lights on as well as the request green lights, 

- the command "M150 B" turns on red lights as well as the requested blue lights.

- the command "M150 R U B" turns all the lights off

- finally the command "M150 R0", "M150 G0", or "M150 U0" turns on all lights making it appear white

I've tried uninstalling/reinstalling the plug-in to no notable change in the actions above and have verified pin location and even moved to other GPIO pins.

chrstrvs, 2020-09-07:
Hi!

I'm experiencing the same issue. Have you had any luck finding a solution?

Regards,
Christian

precision, 2020-09-08:
Are either or both of you using pigpiod?  I hate to admit it, but OctoPrint setup is really outdated.  I'll see if I can get some time later this week to get my Pi upgraded with the latest OctoPrint to reproduce.

chrstrvs, 2020-09-12:
I'm sorry if I have wasted your time, but it turns out I'm an idiot =P
Before wiring a bunch of relays I wanted to try out your plugin and the LED strips powering them directly from the Pi, but it wasn't until now that I realized that the GPIO pins are not grounded to make the LED strip light up, but powered on, so of course it won't work.
I'm a complete beginner when it comes to Pi's and I've never used them for anything else than Octoprint. Can the GPIO pins be connected to ground?